PiSugar/PiSugar #113: I2C Failing Raspbian w/ Zero 2 W

blazejj759, 2023-03-22:
I have currently been trying to connect my Raspberry Pi Zero 2 W with the PiSugar 3 Plus. Everything is wired, and the pi is getting power, however, it is failing to find any of the i2c connections when following your instructions. I will note that under ls /dev/*i2c* there is only i2c-1 and i2c-2. Is this simply an issue with the newest Raspbian release and how the Pisugar is set up? Is there some method of getting around this? I need to get the connections secured so that I can continue with a project due in the next two or three weeks, so any help is greatly appreciated!

spaceman1988, 2023-05-03:
If it helps... I returned my PiSugar 3 and went with an S and gave up on UPS i2c functionality. The HAT I was using, a sound card, would not work in conjunction with the UPS. Trying to reassign i2c to another GPIO didn't work either. The S model has a feature (auto-start on/off switch) to disable the connection to GPIO3 allowing the HAT to have free access to i2c - then everything worked normally.

JdaieLin, 2024-06-05:
> I have currently been trying to connect my Raspberry Pi Zero 2 W with the PiSugar 3 Plus. Everything is wired, and the pi is getting power, however, it is failing to find any of the i2c connections [...]

Raspbian comes with i2c tools by default, but its i2c interface is not enabled by default. You can use this command to see if the i2c works fine:
```
i2cdetect -y 1
```
Pisugar-server's installation script will help you open the i2c interface. It use the default i2c-bus (i2c-1) of the Raspbian. If you are using other system images, you may need to change the i2c bus in the pisugar-server configuration.

closed 2024-07-20 by github-actions[bot]