PiSugar/PiSugar #115: Design Flaw: Power Button on PiSugar Board itself, no pins to extricate external power button

droideggs, 2023-04-10:
One obvious flaw here is that you need to manually press the power button on the PiSugar board itself to turn it on. The side power button will not turn back on the device. Would have been nice to have added pins so that a power switch/button can be added so a device can be turned on without having to press the power button on the board itself. Many projects have enclosures over these Pi projects so getting to the button itself is difficult.

spaceman1988, 2023-04-29:
Same question for the PiSugar S - turning the Pi off is easy, but no way to use the tiny buttons if the devices is in a case. What about the CTRL solder pins?

wlmh110, 2023-05-04:
The button on the board is a reset button, not used to turn on or off the PiSugar 3. To power on with the side button, it requires a short press followed by a long press. If you do not want to use this method to power on, you can disable the anti-misoperation function through I2C or WEBUI.

An external switch can be soldered onto the reserved solder pad on the PCB. Please refer to the following link for specific soldering instructions:
https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series#pcb-instructions

closed 2024-06-22 by github-actions[bot]