PiSugar/PiSugar #116: Please Define PiSugar S CTRL-OFF-ON Pad Usage

spaceman1988, 2023-04-30:
Request a better definition of the PiSugar S CTRL-OFF-ON Pads:

1. Do these pads control power from the PiSugar to the Pi?
2. Do these pads in any way connect to or effect the GPIO pin functions?
3. Do the pads perform the same function as the adjacent small on-off slide switch?
4. If the pad are wired to an external switch does the external switch override the slide switch on the board?
5. If an external switch is connected to the pads should the slide switch on the board be placed in the "on" or "off" position?
6. What rating should an external switch have as a minimum?
7. Can you turn ON the Pi by connecting CTRL and OFF to a momentary pushbutton switch while leaving the slide switch on the board in the "ON" position.

wlmh110, 2023-05-04:
Before answering your questions one by one, let me explain the working principle of the power switch. These three pads and the adjacent slide switch are used to control a load switch to turn on or off the power supply. The CTRL pin is directly connected to the control pin of the load switch. When it is connected to the ON pad, it represents a high level (connected to the battery positive through a 10K resistor), while it represents GND when connected to the OFF pad. Due to the existence of the 10K resistor, connecting to GND will have a main effect.

These interfaces control the power supply from PiSugar to the Raspberry Pi.

These interfaces do not connect to or affect the GPIO pin functions.

These interfaces perform the same function as the adjacent small on-off slide switch, and both can control the power supply.

When these pads are connected to an external switch and used together with the slide switch, any of the switches being closed will turn off the power output.

If using an external switch with a self-locking function, the slide switch should be placed in the "ON" position. If the external switch does not have a self-locking function, the slide switch should be placed in the "OFF" position.

The switch itself does not carry current, so there is no specific requirement for it.

You can turn on the Pi by connecting CTRL and OFF to a momentary pushbutton switch while leaving the slide switch on the board in the "ON" position. This will briefly cut off the power and then restore it, thereby waking up the Raspberry Pi. However, we recommend using the custom switch to wake up the Raspberry Pi, which works by briefly pulling the SCL pin low. This is a recommended way to wake up the Raspberry Pi while keeping the power supply connected.

spaceman1988, 2023-05-08:
I appreciate this very precise answer - exactly what I needed.

The PiSugar S provides many options so following PiSugars advice to use the
'S' was a great solution.

The problem with using the GPIO3 pin was caused by the low power state on
the Pi board. I am using a Pi4B which has a Low Power Mode that leaves on
the 5v line functional - the GPIO3 is not monitored. It took me some time
to realize why momentarily shorting GPIO3 wouldn't generate a restart. I'm
not sure why my Pi4 was defaulted to Low Power Mode but I modified the boot
sequence to instruct the Pi to go into a normal shutdown/halt mode
which preserves GPIO3 functionality. Once I accomplished that I was able to
restart the Pi using a momentary short of GPIO3 to ground. It's nice to
know that if a Pi4 owner wants to use Low Power Mode the PiSugar S provides
that capability through the CTRL-OFF pads.

Thank You.

closed 2024-06-22 by github-actions[bot]