PiSugar/PiSugar #117: Change PiSugar 3 RTC i2c address

amenekowo, 2023-07-02:
PiSugar 3 is amazing to include a rtc compatible with ds1307 driver. But I prefer to use a real DS3231SN module for higher time accuracy. Document says we can change PiSugar's mcu address. So if it's possible to alter the address of built-in rtc?
Any help appreciated ☺️

JdaieLin, 2023-09-18:
@wlmh110

amenekowo, 2024-06-21:
Hi there!
Can we change i2c address or disable the rtc functionality now?
Please let me know if we got any information about it.

JdaieLin, 2024-07-22:
Hi there, PiSugar 3 only support changing the i2c address for battery management (search Custom I2C address on the wiki page). We will consider add the changing support for the rtc i2c address in the coming firmware.

closed 2024-09-06 by github-actions[bot]

nvmd, 2024-11-01:
Hi @JdaieLin,
I'm just wondering was the feature of changing RTC address ever implemented? Should this issue be reopened or a new one created?

It would be great to have this feature, as one of screens for RaspberryPi Zero (namely [Waveshare 2.13inch HAT+ V4](https://www.waveshare.com/wiki/2.13inch_e-Paper_HAT+)) which is very often used in combination with PiSugar, now also includes DS3231 on board, making use of this popular modules together quite problematic.