PiSugar/PiSugar #119: [PiSugar 3 PLUS] Battery charging behavior

Udeste, 2023-07-04:
On PiSugar 3 PLUS the 4 battery charging green leds never stop flashing when the battery is fully charged.
Also the UI and the `get battery_charging` command says that the battery is still charging after it reached 100%.

Is it my unit that is faulty or is a known issue/behavior? The firmware version on board is 1.3.4.

I also noticed that if I manually disable the charging with `set_allow_charging false` , it correctly stop charging, but the leds are still flashing and the battery drains like it is used for powering the raspberry. 
Is it supposed to bypass the battery when the power plug is connected and the battery is fully charged?

JdaieLin, 2023-09-18:
@wlmh110

jayanty, 2024-01-01:
Just got a new 3 PLUS and I seem to have the same issue. Any chance someone here has an update?

JdaieLin, 2024-06-06:
Please refer to this reply
#137

closed 2024-07-21 by github-actions[bot]

BNWEIN, 2025-03-04:
Hi

I have a pisugar3 (not the plus) and i am experiencing this same thing. Its brand new, and has been charging for over 3 hours on a 5v adapter.