PiSugar/PiSugar #122: SCL Wake up is not working

gmagendzo, 2023-09-30:
Hey all, I set up the the 3rd bit of the 0X20 address to 1 by using `i2cset -y 1 0x57 0x20 0x48` in order to enable the SCL awakening function, but it doesn't seem to be working correctly. I disconnected power to the PiSugar3 and shutdown the RPi4, but when I connect power back to the PiSugar3 it doesn't seem to boot the RPi4. Not sure what I'm doing wrong here, and would love some guidance in order to get the functionality working.

JdaieLin, 2024-06-06:
@wlmh110

closed 2024-07-21 by github-actions[bot]

herrfrei, 2024-10-01:
The issue is not solved. Have the same right now also. The register cannot be written (value does not change):

```sh
i2cset -y 1 0x57 0x20 0x88
i2cget -y 1 0x57 0x20
0xc0
```