PiSugar/PiSugar #126: [Question] Can you set PiSugar 3 to boot more than once a day?

phalse, 2023-11-07:
I just go my PiSugar3 working with a RPI Zero WH and want to know if it is possible to set the PiSugar to wake up the Pi more than once a days (web UI only specifies one time per day).  Can you do this though the cli API?

wombat135, 2024-02-09:
I have the same request.

phalse, 2024-02-10:
As a work around solution, I just ended up writing a script that ran at boot. The script queries the Sugar for current time, then adds X hours and sets the alarm for current time + X, then shuts down.

wombat135, 2024-02-10:
I was going to do the same as you. Date+X then shutdown.  As I was playing the Sugar a little more, I realized that when I set an rtc_alarm via the WebUI or via the CLI, it only powers the pi when the pi sugar is completely turned off.  I cannot figure out a way to turn the pi sugar off without using the physical button/switch.

End state is to have a pi that is remote that turns on every couple of hours, does a few tasks, and then turns off.

jacrooke, 2024-06-03:
Have you figured out a way to turn the PiSuagr off yet? Would you mind sharing an example of the script you ended up using for setting an alarm for a future time then shutting down your Pi?

JdaieLin, 2024-06-06:
> As a work around solution, I just ended up writing a script that ran at boot. The script queries the Sugar for current time, then adds X hours and sets the alarm for current time + X, then shuts down.

This method is feasible because RTC can only store one alarm clock, so you can only set the next alarm clock at startup.

closed 2024-07-21 by github-actions[bot]