PiSugar/PiSugar #128: Pi Sugar2 on Rasperry Pi 02W with Rasperry Pi OS Bookworm not working

kagehisa, 2024-01-12:
I recently switched my project from a Pi 0W with 32 bit Buster to a Pi 02W with 64 bit Bookworm. So far everything works fine. The only issue is that the i2c address 0x75 is not showing up and I can not access the battery level. 

`journalctl -u pisugar-server` shows only:
>Jan 12 20:10:32 pwni pisugar-server[10112]: [2024-01-12T19:10:32Z INFO  pisugar_core] Init battery...
Jan 12 20:10:32 pwni pisugar-server[10112]: [2024-01-12T19:10:32Z INFO  pisugar_core] Init battery...
Jan 12 20:10:32 pwni pisugar-server[10112]: [2024-01-12T19:10:32Z INFO  pisugar_core] Init battery...
Jan 12 20:10:32 pwni pisugar-server[10112]: [2024-01-12T19:10:32Z INFO  pisugar_core] Init battery...
Jan 12 20:10:32 pwni pisugar-server[10112]: [2024-01-12T19:10:32Z INFO  pisugar_core] Init battery...
Jan 12 20:10:32 pwni pisugar-server[10112]: [2024-01-12T19:10:32Z INFO  pisugar_core] Init battery...
Jan 12 20:10:32 pwni pisugar-server[10112]: [2024-01-12T19:10:32Z INFO  pisugar_core] Init battery...
Jan 12 20:10:32 pwni pisugar-server[10112]: [2024-01-12T19:10:32Z INFO  pisugar_core] Init battery...
Jan 12 20:10:33 pwni pisugar-server[10112]: [2024-01-12T19:10:33Z INFO  pisugar_core] Init battery...
Jan 12 20:10:33 pwni pisugar-server[10112]: [2024-01-12T19:10:33Z INFO  pisugar_core] Init battery...

`i2cdetect -y 1` shows:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- 32 -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

Any pointer as to what is wrong?

josh10111, 2024-02-29:
I'm having this same exact issue with the same results from `journalctl -u pisugar-server` and `i2cdetect -y 1`.

JdaieLin, 2024-06-05:
It might happen when pi0w's usb port is connecting to a external power source of PC. In this case the battery board detects it as the shutdown status(no demand for battery power) and the chip (0x75) will turn off automatically.

closed 2024-07-20 by github-actions[bot]