PiSugar/PiSugar #134: Installation version window

gabrielb357, 2024-03-06:
When I try to pick pisugar version in the window pop up during installation it doesn't allow me to navigate or select anything. I looked for similar posts and I'm sure this could potentially just be common sense but wasn't sure where else to go

gabrielb357, 2024-03-07:
I guess the installation worked well enough for me to get it to work and with a display working, regardless of not being able to completely finish the installation lol. The only problem is that i cant get a power management ui up now.

nonik0, 2024-03-11:
I also had this issue. When dpkg runs as part of the install script, I was unable to interact with the menu to choose the PiSugar hardware type. I was able to work around this by manually running dpkg myself and the UI worked for me then. E.g.

```
sudo dpkg -i /tmp/pisugar.*/pisugar-server_1.7.5_arm64.deb
sudo dpkg -i /tmp/pisugar.*/pisugar-poweroff_1.7.5_arm64.deb
sudo dpkg -i /tmp/pisugar.*/pisugar-programmer_1.7.5_arm64.deb
```

gabrielb357, 2024-03-12:
> I also had this issue. When dpkg runs as part of the install script, I was unable to interact with the menu to choose the PiSugar hardware type. I was able to work around this by manually running [...]

Thanks for the response! I tried running these but just says file or directory not found? I'm kind of new to all this so just learning as I go. I actually am able to access the web ui but it just shows sugar 2 and rtc wont update properly

elkay, 2024-03-13:
Having the same issue. Those files aren't in the tmp folder.

elkay, 2024-03-13:
Figured it out, you need to ls the /tmp directory and get the string that represents "*" in the path. Replace "*" with that string and the scripts will run and you can navigate the menu.

> Thanks for the response! I tried running these but just says file or directory not found? I'm kind of new to all this so just learning as I go. I actually am able to access the web ui but it just [...]

gabrielb357, 2024-03-14:
> Figured it out, you need to ls the /tmp directory and get the string that represents "_" in the path. Replace "_" with that string and the scripts will run and you can navigate the menu.

my bad i just realized. I never mentioned i'm using my pisuagr as a pwnagotchi. I feel pretty dumb for barely realizing. Although, I do appreciate the help

gabrielb357, 2024-03-27:
I was able to unintentionally figure this ojt myself. I was installing packages for another pwnagotchi plugin and it prompted me to install the right pisugar firmware like it just picked ip where i left off:

Apt-get update

apt-get install libcurl4-openssl-dev libssl-dev

JdaieLin, 2024-06-06:
To avoid the version selection issue please try using this command to install:
```
wget https://cdn.pisugar.com/release/pisugar-power-manager.sh
bash pisugar-power-manager.sh -c release
```

closed 2024-07-21 by github-actions[bot]