PiSugar/PiSugar #137: Battery NEVER stops charging? (LED lights NEVER stop blinking)

Dochartaigh, 2024-04-04:
**Note:** This repo is mainly about the entire PiSugar project, cases and wiki. We usually process these issues once a month. For quick response about PiSugar products, please contact pisugar.zero@gmail.com. For software issues, please submit your issuse to https://github.com/PiSugar/pisugar-power-manager-rs/issues

PiSugar 3, 1200mah, bought from Amazon

The battery seems to NEVER stop charging? (LED lights NEVER stop blinking). Even when it's detached from the Raspberry Pi (which I don't know if I HAVE to install special software on the Raspberry Pi Zero 2 W itself to make it stop charging?). I let it charge on a known-working USB-C cable/charging-brick and 20+ hours and those LED's are STILL blinking!

Am I doing something wrong? Any fix?

JdaieLin, 2024-06-05:
Hi Dochartaigh, it won't stop blinking is the cable is connected and actually it's not in charging status. We will try fixing this bug in the future version. Thanks for your feedback!

closed 2024-06-05 by JdaieLin

drego85, 2025-11-08:
Hello, the problem persists.

Are there any updates on when it will be resolved?

Thanks 

PS. Another thread: https://github.com/PiSugar/PiSugar/issues/119