PiSugar/PiSugar #141: How to connect a PiSugar 3 to a Pi Zero 2W without mounting them directly to each other?

Manish181992, 2024-05-28:
Hi,

I want to mount my Pi Sugar 3 separate from my Pi Zero 2W. How do I connect the two? My guess is to use the extension interface on PiSugar 3, but what type of connector does it use?

**Note:** This repo is mainly about the entire PiSugar project, cases and wiki. We usually process these issues once a month. For quick response about PiSugar products, please contact pisugar.zero@gmail.com. For software issues, please submit your issuse to https://github.com/PiSugar/pisugar-power-manager-rs/issues

JdaieLin, 2024-05-28:
connect the extension interface to pizero:
5V Output -> Pi 5V
GND -> Pi GND
SDAT -> Pi I2C1 SDA (GPIO2)
SSCL -> Pi I2C1 SCL (GPIO3)

closed 2024-05-28 by JdaieLin

Manish181992, 2024-05-28:
What type of connector can I use for the extension interface?

JdaieLin, 2024-06-06:
6Pin with 1.75mm pitch