PiSugar/PiSugar #145: support for pi 5 for 5A max output

beitong95, 2024-07-19:
**Note:** This repo is mainly about the entire PiSugar project, cases and wiki. 
We handle issues via email. If you have any questions about PiSugar products, please contact us at pisugar.zero@gmail.com and we will respond promptly. For software code issues, please submit your issuse to https://github.com/PiSugar/pisugar-power-manager-rs/issues

JdaieLin, 2024-07-22:
Hi there, PiSugar3 Plus can provide a maximum current output of 3A, which mainly depends on the output capacity of the current lithium battery and the maximum current allowed by the wire. If you want to achieve the output requirement of 5A, you may have to reduce the voltage by reducing the high voltage, which will be a completely new circuit design.

JdaieLin, 2024-07-22:
It is difficult for lithium batteries in portable devices to meet this requirement.

closed 2024-07-23 by beitong95