PiSugar/PiSugar #148: PiSugar 3 Software Power Off Doesn't Fully Power Off System

LouisLenoir4, 2024-08-21:
**Description:**
I am using a PiSugar 3 with a Raspberry Pi 4 and I'm experiencing an issue with the software power off feature. When I click the power off button, the Raspberry Pi 4 shuts down, but the PiSugar 3 battery keeps powering the system.
The LEDs are still emitting light, and the fan is still running, even though the Raspberry Pi 4 is powered off. Additionally, I am not able to restart the Raspberry Pi 4 using the buttons on the PiSugar 3.
I have updated the PiSugar server and firmware to the latest version (1.7.7), but the issue persists.
**Environment:**

PiSugar 3
Raspberry Pi 4
PiSugar server and firmware version 1.7.7

Steps to Reproduce:

Connect the PiSugar 3 to the Raspberry Pi 4.
Set the power off mode to "soft poweroff"
Click the power off button on the PiSugar 3.
Observe that the Raspberry Pi 4 shuts down, but the PiSugar 3 battery continues to power the system.
Attempt to restart the Raspberry Pi 4 using the PiSugar 3 buttons.

Expected Behavior:
When the power off button is clicked, the entire system (Raspberry Pi 4 and PiSugar 3) should power off completely, and I should be able to restart the Raspberry Pi 4 using the PiSugar 3 buttons.
Actual Behavior:
The Raspberry Pi 4 shuts down, but the PiSugar 3 battery continues to power the system, with the LEDs and fan still running. I am unable to restart the Raspberry Pi 4 using the PiSugar 3 buttons.
Workaround (if any):
I have not found a reliable workaround for this issue.
Additional Information:
The hardware power off feature works as expected, but it is too dangerous for the databases on my device, so I need to use the software power off feature.

JdaieLin, 2024-10-04:
It seems to be an issue of pisugar-poweroff.
@fengyc

JdaieLin, 2024-10-07:
If you directly use the shutdown command to turn off the Pi4, will the PiSugar also shut down afterward?

JdaieLin, 2024-10-07:
Is this a consistently reproducible issue or does it occur sporadically?

JdaieLin, 2024-10-07:
If you encounter this issue again, please paste the OS log here. After booting up, use journalctl -u pisugar-poweroff -n 100 to check if there were any I2C errors before the last shutdown.

closed 2024-11-21 by github-actions[bot]