PiSugar/PiSugar #149: Remote power button with 5v LED

thepctech, 2024-09-05:
I bought a LED backlit power button in hopes that I could remotely power on and off the device connected to a pi 3.  The sugar 3 and pi will be sealed in a case and I was hoping to mount the button to the outside of the case and see if the pi is being powered or if the pi is off, if off I'd like to press the button to turn it on.  This is a standard 4 pin switch (2 for LED), and (2 for momentary on).

Is this possible from the pins on the pi sugar 3 plus, or are there pads that I can solder to that could provide this functionality?  One last thing I see some people say that you need a resistor on one of the LED legs, is this necessary? 

The wiki is not clear if this is possible.

Thanks, 
Brian

thepctech, 2024-10-10:
I have not figured this out either.  A diagram as to where to connect the wires, to which
 pads would be great!!

JdaieLin, 2024-10-10:
https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series#pcb-instructions
power button should be connect to 2(BAT) and 10(PWR)

mikepaxton, 2024-11-03:
Regarding the original posters question, connecting pad 2 and pad 10 works with a momentary switch.  

I'm using a 4 pin button with LED, but mine is a constant on/off switch.  As a workaround I mounted the PiSugar 3 separate from the Pi 3A+ and spliced the on/off button in between the 5v and ground coming off the pogo pins.  I soldered all of the wires (including i2c) to the top of each pogo pin.  I used a couple pairs of JST-PH connectors, 5v and GND out to the button and another pair of JST connectors for 5v and GND into the Pi 3A+.  

When the button is pressed, the LED turns on and the Pi boots.   To turn the Pi off, I use the standard sudo shutdown -h now and once the Pi has fully shutdown I can press the button once more to power it off fully, turning off the button LED.

Because this Pi is headless and inside a case it makes it difficult to determine when the Pi has fully shutdown.  To get around this I installed a second (green) LED in the case and tied it to GPIO4.  In the /boot/firmware/config.txt I added the following: gpio=4=op,dh

This turns the green LED on once the Pi starts loading the Linux kernel and turns it off once the Pi has fully shutdown.  At that point, I can press the power button to completely cut the power to the Pi.

I hope this helps for anyone trying to do something similar.

closed 2024-12-19 by github-actions[bot]