PiSugar/PiSugar #153: How to make the scheduler work?

rnaud, 2024-11-07:
Hello team,

I am using the web interface to create a wake up schedule. I set the time up, then I used either:

`pisugar-poweroff --model "PiSugar 2 (2 LEDs)"
or 
`sudo shutdown -h now`

But the device never boots up again. 

A few questions or ideas
* Is the time I'm setting using the same timezone as mine? 
* Anyway for me to check via a CLI that the RTC wakeup alarm is properly set? 
* Anything else I can do to troubleshoot?

JdaieLin, 2024-11-14:
The hardware rtc has no timezone actually, it's using your timezone in software. If you can see the time up after refreshing the webpage it's properly set. As for the failure to boots up again please contact our product support for more information: pisugar.zero@gmail.com

closed 2024-11-14 by rnaud

rnaud, 2024-11-14:
The daily reboot did work, it was just not lined up with my current timezone