PiSugar/PiSugar #154: Pisugar-server 'Invalid'

Zanop, 2024-11-13:
Hi,

Looking at syslog I am seeing this constantly being printed:

Nov 13 13:30:27 Curigotchi pisugar-server[243]: [2024-11-13T12:30:27Z ERROR pisugar_server] Invalid
Nov 13 13:30:41 Curigotchi pisugar-server[243]: [2024-11-13T12:30:41Z ERROR pisugar_server] Invalid
Nov 13 13:30:53 Curigotchi pisugar-server[243]: [2024-11-13T12:30:53Z ERROR pisugar_server] Invalid
Nov 13 13:31:04 Curigotchi pisugar-server[243]: [2024-11-13T12:31:04Z ERROR pisugar_server] Invalid
Nov 13 13:31:15 Curigotchi pisugar-server[243]: [2024-11-13T12:31:15Z ERROR pisugar_server] Invalid
(END)

JdaieLin, 2024-11-14:
Does the webUI work properly?

Zanop, 2024-11-14:
I don’t see errors in the ui

JdaieLin, 2024-11-14:
Yes the webUI will keep querying the server. The server will resonse "Invalid" if there's an invalid command. I'm not sure which query is causing this, maybe it's dereapted in the server but still left in the UI. I will check it later.

JdaieLin, 2024-11-15:
> Hi,
> 
> Looking at syslog I am seeing this constantly being printed:
> 
> Nov 13 13:30:27 Curigotchi pisugar-server[243]: [2024-11-13T12:30:27Z ERROR pisugar_server] Invalid Nov 13 13:30:41 Curigotchi pisugar-server[243]: [2024-11-13T12:30:41Z ERROR pisugar_server] Invalid Nov 13 13:30:53 Curigotchi pisugar-server[243]: [2024-11-13T12:30:53Z ERROR pisugar_server] Invalid Nov 13 13:31:04 Curigotchi pisugar-server[243]: [2024-11-13T12:31:04Z ERROR pisugar_server] Invalid Nov 13 13:31:15 Curigotchi pisugar-server[243]: [2024-11-13T12:31:15Z ERROR pisugar_server] Invalid
> 
> No further information is seen as to what is invalid.
> 
> Seems to be casued by my browser being open at 'http://10.0.0.2:8421/#/".
> 
> Regards, Vlado

Which model are you using, PiSugar 2 or 3?

JdaieLin, 2024-11-24:
Try set up a schedual wake up time and see if the 'Invalid' log comes up again.

closed 2025-01-08 by github-actions[bot]