PiSugar/PiSugar #156: power off question

mtlmaks, 2024-11-29:
Hi, I have a raspberry pi 4 on a pisugar 3 board with its battery, with the pisugar and the rpi both connected to 5v power. When no power is connected and I shut down the rpi under rpi os, it shuts completely down, i.e. not lights anymore on the rpi. This way, when reconnecting power to the boards, the rpi starts up as the wake up setting in the pisugar power managment settings is set to power restore.
Now, when I run the rpi under [Volumio OS](https://volumio.com/get-started/), which is based on rpi os, the rpi does not completely shut down when no power is connected, with still a red light on on the rpi, so it still draws power from the battery as it seems. This also prevents the rpi to start up when power is restored.  Any idea how to solve this, either how to completely shut down the rpi under volumio, or how to have it reboot when power is restored? Any help would be highly appreciated! Thanks! :)

JdaieLin, 2024-12-10:
Had you pisugar-poweroff together with pisugar-server on your OS?

mtlmaks, 2024-12-11:
Hi, I'm not sure what you mean by installing pisugar-poweroff... I installed the `pisugar-power-manager`... But like I said, the pi4 shuts down completely under `Raspbian/Raspberry Pi OS`, but not under Volumio. This seems to be the same issue as in this post: https://github.com/PiSugar/PiSugar/issues/148

JdaieLin, 2024-12-12:
You can find the pisugar-poweroff package here: [pisugar-poweroff_1.7.8-1_armel.deb](https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-poweroff_1.7.8-1_armel.deb). It's a plugin to make sure the shut down completely.

mtlmaks, 2024-12-12:
Thanks. I tried that, installed https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/v1.7.7/pisugar-programmer_1.7.7-1_armhf.deb, rebooted, but the rpi still doesn't shut down completely under Volumio... I did get a warning at the end of the install progress, that raspi-config wasn't found, which does not exist under Volumio, not sure if that could have an impact...

mtlmaks, 2024-12-13:
Oh snap, I just noticed that I installed the wrong package... Sorry about that, will try again and report back.

JdaieLin, 2024-12-13:
It should access the raspi-config to enable i2c interface, but raspi-config only exist on Raspbian. In this case you should find a  way to enable i2c on Volumio OS, then make sure which i2c bus the PiSugar is using on your system. To custom the i2c bus you can modify in /etc/pisugar-server/config.json

JdaieLin, 2024-12-13:
you can run
```
 ls /dev/i2c*
```
to check the i2c bus list on Volumio

mtlmaks, 2024-12-13:
Hi, first of all, thanks for trying to help me :) 
The i2c bus seems to be set correctly as far as I can see, not sure however if `"i2c_addr"` should be null...:

[image]

JdaieLin, 2024-12-21:
Had you choose the pisugar model during the pisugar-server installation? Some config values depends on the model.

JdaieLin, 2024-12-22:
I guess the installation of pisugar-server on Volumio OS didn’t go very smoothly, which resulted in some missing parameters.

closed 2025-02-04 by github-actions[bot]