PiSugar/PiSugar #157: the blue led won't turn off

terminaldweller, 2025-01-22:
The blue led won't turn off on my pi sugar 3. What does that mean?
How can I try to further investigate the problem?

furynick, 2025-02-07:
The blue LED means power is provided to Pi board.
I don't know if it's the same issue but my PiSugar3 doesn't shutdown from time to time when Pi OS shuts down and I have to manually power off then power on PiSugar. Power cycle then work during a couple of days.
Power manager is installed, battery status is retrieved every minute, power on is scheduled at 6am every day and OS is shutdown at 10pm.
PiSugar is powered by a solar charger USB outlet.

JdaieLin, 2025-02-14:
> The blue LED means power is provided to Pi board. I don't know if it's the same issue but my PiSugar3 doesn't shutdown from time to time when Pi OS shuts down and I have to manually power off then [...]

The script for installing power-manager also includes the installation of the pisugar-poweroff service, which is specifically responsible for sending the shutdown command to PiSugar when the Raspberry Pi system powers off. It sounds like there might be a bug in this part. You can try updating the firmware first: https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series#%EF%B8%8F-ota-firmware-upgrade

JdaieLin, 2025-02-14:
> The blue LED means power is provided to Pi board. I don't know if it's the same issue but my PiSugar3 doesn't shutdown from time to time when Pi OS shuts down and I have to manually power off then [...]

Sometimes, the Raspberry Pi’s LED turning off doesn’t necessarily mean it has completely stopped working. If you use the shutdown command to power off, try waiting for two to three minutes to see if the PiSugar powers off properly.

furynick, 2025-02-14:
The pi is on another place, it stayed several days before I can manually power cycle PiSugar.
I'll try firmware update.

furynick, 2025-03-17:
No more hang since firmware upgrade, problem solved from my pov.

closed 2025-03-19 by JdaieLin