PiSugar/PiSugar #158: Y cable issue

chrsturbo, 2025-01-23:
Trying to have both the sugar and the p0 connected with a Y seems to cause the sugar to charge from the p0 is there a way to prevent this?

JdaieLin, 2025-01-31:
If the Y cable can also provide power to the Raspberry Pi, it might cause confusion. I'd like to understand what usage scenario this Y cable is intended for. Also, which PiSugar model are you using?

chrsturbo, 2025-01-31:
It's for a teslausb device which needs to be connected to the car over the
data port but I want the sugar to give it more life than the car will
provide. I got a C/micro Y which does data (rare) to allow both the PI and
the sugar to connect to the cars A. Pi 0 2W with the 3rd gen sugar.
-Chris
PS https://github.com/marcone/teslausb

JdaieLin, 2025-02-03:
The key issue here is that the PiSugar and PiZero cannot be connected to the same USB power source simultaneously. Therefore, I recommend making a special modification to the Y cable: cut the two power wires in the microUSB connector going to the PiZero, leaving only the two data wires intact.

chrsturbo, 2025-02-03:
cool thanks!

chrsturbo, 2025-02-03:
Gonna try this dodad
https://elabbay.myshopify.com/products/usb3-bm-bf-v1a-micro-usb3-0-type-b-male-to-micro-usb3-0-type-b-female-pass-through-breakout-board?_pos=3&_sid=d492d5d9a&_ss=r&variant=45423382147

closed 2025-02-20 by JdaieLin