PiSugar/PiSugar #159: PiSugar 3 ... me.local:8421 say "...%"

a6k, 2025-02-19:
i see a white battery (logo) with "...%" 
the rest of the page is ok. i can change settings and the ios-app i sconnecting but says "0%" too

JdaieLin, 2025-02-20:
It looks like the issue is caused by poor contact between the pogo pins and the Raspberry Pi's I2C pins. You can try following the solution for the first FAQ question and then reinstall the battery.

https://github.com/PiSugar/PiSugar/wiki/FAQ#1-why-does-the-raspberry-pi-remain-unpowered-after-the-pisugar-is-turned-on

a6k, 2025-02-20:
yes, thank you ! that fixed the issue :)

closed 2025-02-20 by a6k

mscrnt, 2025-03-01:
I hate to resurrect this, but I am having the same issue. I tried cleaning the contact between the pogo pins but that didn't resolve the issue.

mscrnt, 2025-03-02:
Ok, that's what I am suspecting is my issue. I'll probably end up resoldering those points to allow the pins to contact better. Thanks