PiSugar/PiSugar #161: Sugar 3 powering on Pi when it’s charging.

BNWEIN, 2025-03-01:
Hi

I have a pi sugar 3, and a raspberry pi 2w

When the pi and sugar are both off, I plug the a USB C cable into the sugar to charge it, and it powers the Pi on. Then I can’t turn it off unless I unplug it from charging 

How do I set it so when I plug in to charge if the pi is already off it should stay off?

JdaieLin, 2025-03-03:
Hi 

did you enable the "On Power Restore" option in the Schedule Wake Up function? This may cause the PiSugar to automatically start the Raspberry Pi when charging. As for the issue you mentioned, where it cannot be turned off unless the power is disconnected, this is quite unusual. You might need to record a video and contact our product support via email so we can confirm whether it's a product defect.

BNWEIN, 2025-03-03:
Hey

After installing power manager it now works as expected! This can be closed. Thanks

closed 2025-03-03 by BNWEIN