PiSugar/PiSugar #162: [Pisugar 3] Battery not charging

max1mvs, 2025-03-14:
I have a brand new Pisugar installed on a Raspberry Pi® Zero 2 WH running Raspberry Pi OS Lite (32‑bit). I installed all the relevant files for it and can get the server up and running. But the battery seems to constantly be at 0% while the green lights continually flash.
I cannot get any information from the shell, and I keep running into errors saying "Invalid request." when trying commands such as `echo "get battery" | nc -q 0 192.168.x.x 8423`.  Trying to update the firmware using `curl https://cdn.pisugar.com/release/PiSugarUpdate.sh | sudo bash` also returns errors:

`  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1553  100  1553    0     0   2300      0 --:--:-- --:--:-- --:--:--  2304
--2025-03-14 10:16:04--  https://cdn.pisugar.com/release/PiSugar3Firmware/fm33lc023n/pisugar-3-application.bin
Resolving cdn.pisugar.com (cdn.pisugar.com)... 47.246.48.224, 47.246.48.225, 47.246.48.226, ...
Connecting to cdn.pisugar.com (cdn.pisugar.com)|47.246.48.224|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 23100 (23K) [application/octet-stream]
Saving to: ‘/tmp/pisugar-update.HbNT36v/pisugar-3-application.bin’

/tmp/pisugar-update.HbNT36v/pisugar-3-application.bin        100%[===========================================================================================================================================>]  22.56K  --.-KB/s    in 0.03s   

2025-03-14 10:16:04 (897 KB/s) - ‘/tmp/pisugar-update.HbNT36v/pisugar-3-application.bin’ saved [23100/23100]

Stoping pisugar-server...
[2025-03-14T09:16:04Z INFO  pisugar_programmer] WARNING:
[2025-03-14T09:16:04Z INFO  pisugar_programmer] 1. PLEASE CONFIRM THAT THE BATTERY IS FULLY CHARGED
[2025-03-14T09:16:04Z INFO  pisugar_programmer] 2. SYSTEMD SERVICE pisugar-server MUST BE STOPPED
[2025-03-14T09:16:04Z INFO  pisugar_programmer] OTHERWISE UPGRADE MAY NOT SUCCEED!!!
[2025-03-14T09:16:04Z INFO  pisugar_programmer] CONFIRM? (y or n): 
[2025-03-14T09:16:04Z INFO  pisugar_programmer] 
[2025-03-14T09:16:04Z INFO  pisugar_programmer] Firmware size: 23100
[2025-03-14T09:16:04Z INFO  pisugar_programmer] PiSugar bootloader/bootapp not ready, please reset or wait, retry...
[2025-03-14T09:16:04Z INFO  pisugar_programmer] PiSugar bootloader/bootapp not ready, please reset or wait, retry...
[2025-03-14T09:16:04Z INFO  pisugar_programmer] PiSugar bootloader/bootapp not ready, please reset or wait, retry...
[2025-03-14T09:16:04Z INFO  pisugar_programmer] PiSugar bootloader/bootapp not ready, please reset or wait, retry...
[2025-03-14T09:16:05Z INFO  pisugar_programmer] PiSugar bootloader/bootapp not ready, please reset or wait, retry...
[2025-03-14T09:16:05Z INFO  pisugar_programmer] PiSugar bootloader/bootapp not ready, please reset or wait, retry...
[2025-03-14T09:16:05Z INFO  pisugar_programmer] PiSugar bootloader/bootapp not ready, please reset or wait, retry...
[2025-03-14T09:16:05Z INFO  pisugar_programmer] PiSugar bootloader/bootapp not ready, please reset or wait, retry...`

[image]

How do you troubleshoot this?

Cheers,
Massimo

JdaieLin, 2025-03-14:
Hi Massimo,

looks like the i2c is not connect so there's not data and cannot be upgrade. 
You can use `i2cdetect -y 1` to check if there's i2c device. If not please try following the first FAQ:
https://github.com/PiSugar/PiSugar/wiki/FAQ#1-why-pisugar-is-not-able-to-power-raspberry-pi-why-pisugar-23-cannot-be-detected-on-i2c-bus

max1mvs, 2025-03-14:
I appreciate your help.

I tried to scrape a bit on the solder points and reattached the Pisugar 3. But that didn't work. I can get the battery to start and run the Raspberry Pi, but the problem is the same. Using `i2cdetect -y 1` I get:

[image]

Cheers,
Massimo

max1mvs, 2025-03-16:
Is there any way to troubleshoot this and determine if the unit is defective so I may return it to where I bought it for a replacement?

JdaieLin, 2025-03-19:
Hi Massimo,

Since you have already tried but I2C is still not working, it could be an issue with the PiSugar pins. You should contact our product support via email to check if a product replacement is needed.

max1mvs, 2025-03-20:
Thanks! I tried a couple of more times to "clean" the soldering at the back of the Pi, and now I was able to upgrade the firmware successfully. Does this seem correct now?

[image]

closed 2025-03-20 by max1mvs