PiSugar/PiSugar #174: Replaced battery, how to tune pisugar 2 firmware?

bastionbot, 2025-07-02:
Hi! I replaced the stock 1200mAh battery with a 3700mAh battery, but the firmware on my PiSugar 2 is still using the discharge curve from the smaller battery and powering off early. Is there a way to modify the register so I can get accurate readings?

bastionbot, 2025-07-03:
After some digging, it looks like modifying ip5209.rs and recompiling pisugar-server might be the way to go.

Stock 1200mAh curve:
```
pub const BATTERY_CURVE: [BatteryThreshold; 10] = [
    (4.16, 100.0),
    (4.05, 95.0),
    (4.00, 80.0),
    (3.92, 65.0),
    (3.86, 40.0),
    (3.79, 25.5),
    (3.66, 10.0),
    (3.52, 6.5),
    (3.49, 3.2),
    (3.1, 0.0),
];
```

Modified 3700mAh curve for https://www.amazon.com/dp/B08215B4KK:
```
pub const BATTERY_CURVE: [BatteryThreshold; 10] = [
    (4.20, 100.0),
    (4.05, 95.0),
    (3.98, 80.0),
    (3.90, 65.0),
    (3.82, 40.0),
    (3.72, 25.0),
    (3.55, 10.0),
    (3.35, 5.0),
    (3.10, 2.0),
    (2.75, 0.0),
];
```

So far it seems to be working?

bastionbot, 2025-07-13:
@santhosh-rao It's in the previous comment. You just need to clone the pisugar power manager repo and make some changes, then compile the binaries with.

`git clone https://github.com/PiSugar/pisugar-power-manager-rs`
Use your preferred editor to modify pisugar-power-manager-rs/pisugar-core/src/ip5209.rs

JdaieLin, 2025-07-21:
After you get the new discharge curve, You can add the following to the PiSugar config file (/etc/pisugar/config.json): "battery_curve": [[3.2, 10], [3.5, 60], [3.8, 90]]
to use a custom battery voltage-battery level curve.

bastionbot, 2025-07-21:
Oh, awesome. That's much simpler!

closed 2025-07-21 by bastionbot